robottini/grbl-servo #1: Motor never stop

eatyone, 2016-03-09:
Hi, many thnx for your work, but i've got a problem... Motor start with M03 S128, but never stop... Any idea?

Best Regards, 

Guillaume

robottini, 2016-03-09:
You can change this values in the file **spindle_control.c**:
define RC_SERVO_SHORT 15 // Timer ticks for 0.6ms pulse duration (9 for 0.6ms)
define RC_SERVO_LONG 32 // Timer ticks for 2.5 ms pulse duration (39 for 2.5ms)

and put 
define RC_SERVO_SHORT 9 // Timer ticks for 0.6ms pulse duration (9 for 0.6ms)
define RC_SERVO_LONG 39 // Timer ticks for 2.5 ms pulse duration (39 for 2.5ms)

I hope it works.

eatyone, 2016-03-09:
Soorrrry for the issue... There is no issue, all is fine... I just forgot to check my servo... It was a continuous one :( :( Have to check with a real servo. Many thanks for ur hack anyway!

closed 2016-03-10 by robottini