robottini/grbl-servo #3: Why this is not a fork?

soswow, 2016-03-19:


robottini, 2016-03-19:
This should have been a fork, but for my mistake it isn't a fork.,

soswow, 2016-03-20:
I would like to compare what exactly you've changed and added. At least I would like to know what commit # you based on your first changes.

robottini, 2016-03-20:
I changed only the file spindle_control.c.

soswow, 2016-03-20:
Sure. But what was the git commit hash you used as a base? (this file was
probably changed since you took it from master)

robottini, 2016-03-20:
#define GRBL_VERSION "0.9i"
#define GRBL_VERSION_BUILD "20150329"

token47, 2016-09-14:
I tried to pull a version of this date from the official grbl repo but I just can't find the right one. Even trying all the commits for this day I still get more changes overall and also more changes in this file.

I also would like to study exactly what was changed, like @soswow asked. Do you have the hash number of the specific commit?

One of the goals is to merge it with newer versions.

token47, 2016-09-14:
I checked for the vanilla tree of this fork on grbl repo again and just cannot make sense of it. The date inside the grbl.h file suggests the file was pulled on Jun-20-2015. This day have only one commit and it is the only commit that sets this date on this file.

But the diff shows a lot of changes so it must not be it. The day right before have a change on the README sugesting the release of 0.9i but the date inside grbl.h is (as expected) Jun-18-2015 and although the diff does not show so much, it still show a lot.

I have tried diffing against this 3 commits (around day 18-20 Jun 2015):

81505e6a8111dd0fcbd4c8c0e88b5c65de5b4904
e6db56451118718fb9ef83624cda6e6ab0ab6ea5
185de028361969e165df82620659711186d07279

Can you remember what was the specific commit?

token47, 2016-09-14:
Found it. It's commit:

Restore parameters minor bug fix.
chamnit committed on 25 Jun 2015

cc38ae2471c756d3c02b11fff82166fa80d80823

If you look at commits before this one until day 20, you can see that he changed the date only on one of two places, misleading the reader. Then, in the next commit after this one he fixes both again, but in this commig the dates were "Jun-25-2015" on the main directory README.md and "Jun-20-2015" on the grbl.h file. But, as @robottini  erased the main directory and commited to his repo only the grbl subdir, then it got harder to find that mistake.

@soswow you may be still interested on this.

robottini, 2016-09-15:
Sorry, I have made a mistake not forking correctly GRBL. Sorry for that.
The only changes are in the file spindle_control.c. I replaced all the
original code with the new code for the servo.

token47, 2016-09-17:
Hi @robottini. No problem, don't worry. I could isolate exactly your changes. Thank you for this port!!

As a side note, the file eeprom.c on that commit I posted above (cc38ae247...) is changed by the original author. Somehow he changed the lineends on that file from LF to CRLF. Git does not care about that but when you try to diff the servo fork to the original one, it appears that @robottini changed the whole eeprom.c file which is not the case. Just run a "dos2unix" on the eeprom.c file from the original tree before the diff and you will be left with only the @robottini `s changes.

I will try to port them to the last version and maybe make them play nicely with the spindle code si I can submit it to the original project. Will leave a note here if I do that.

Thank you again!!!

mattholl, 2017-06-17:
Hi @token47 did you ever manage to isolate the changes to to spindle_control.c? I'd like to be able to use a more recent version of Grbl but the file looks to have changed quite a lot by the v1.1 release? Thanks!

token47, 2017-06-19:
Unfortunately no, I hacked them to a later version (still before 1.x) and never made something publishable, sorry. But I think this have a much higher chance of being done by the original project now that it have a "mega" branch.