robottini/grbl-servo #4: I can not use the grbl controller

atakanzey, 2016-04-21:
I upload the codes correctly. But I can not send the G code because there is no connection the grbl controller (3.5.1) and arduino. How I can fix it. If I use another software to connect arduino for sending G code. Please help me Thank you.

robottini, 2016-04-22:
There are many sw for the connection between PC and Arduino. You can use the sw you prefer. The GRBL controller is only the sw I use, but it is not mandatory

atakanzey, 2016-04-22:
OK. But I want use the GRBL controller 3.5.1. When I select the COM 3 Port in the GRBL controller , it says me "No data from COM port after connection.Expecting Grbl version string." How I can fix that problem ? thank you...

boundlessmaking, 2017-02-22:
had the same problem, figured it out after hours of trying and almost giving up... made an account to show others what to do: 
set the BAUD rate to 115200 in the software when connecting to the arduino with universal gcode sender or grbl-controller. mine was set to 9600 wich was the problem.
[image]

maneshian, 2017-07-09:
I do have the same problem.  Tried so many things, still not working!

LonghornHDTV, 2017-11-23:
Did do you use GRBL v0.9?

Rafeeu, 2018-08-06:
OK. But I want use the GRBL controller 3.5.1. When I select the COM 3 Port in the GRBL controller , it says me "No data from COM port after connection.Expecting Grbl version string." How I can fix that problem?
I use GRBL v0.9

mrtwizta, 2019-03-28:
I have the same problem, tried the At mega 2560 and an MKS Gen2-Z (2560 chip)
"No data from COM port after connection.Expecting Grbl version string."
So I loaded A Marlin printer sketch i use for my 3Dprinter and it works fine, obviously till the heater hardware reports an error.
Just to confirm/prove the hardware and wiring is correct, i can move and home all axis without problems
so it seems the devs need to do some soul searching there?

jtmunson, 2019-07-21:
Having the Correct Com# is essential/ Once set, for reiteration thanks, Boundlessmaking, the baud rate of 115200 work for me. Default was 9600. But 115200 works.

Legeriana48, 2019-11-04:
(Ctrl X)
No data from COM port after connection.Expecting Grbl version string
... not run?

phuong140587, 2020-03-13:
Can I help me? No data from COM port after connection.Expecting Grbl version string
arduino 1.8.12; GRBL servo master,

robin-gdwl, 2020-03-13:
> Can I help me? No data from COM port after connection.Expecting Grbl version string
> arduino 1.8.12; GRBL servo master,

@phuong140587 

have you set the correct com port? 
did you change the baud rate ?

phuong140587, 2020-03-13:
upload complete, 
[image]
[image]
I used arduino Portable 1.8.12 before to upload normally. then I switched to use arduino 1.8.12 installer version, this error occurred. baurd rate: 115200, com5. In the port monitor there is no GRBL response: 0.9 [i.
I loaded another code to re-test the x, y, and servomotor motors still ran normally
I do not speak English well,

PABXStu, 2020-11-14:
Brilliant, Thank You - set baud rate to 115200 connected straight away.

mrtwizta, 2020-11-16:
Pleasure

ueline, 2021-08-16:
@phuong140587 
I have the same problem, did you manage to solve this issue? 
I do not get any respond from the serial monitor. 
I have the following equipment: 
-	Arduino Mega 2560 R3
-	CNC Shield with 2x A4988 Stepper Motor Driver
May I need to change something in the config.h file? 

Thank you so much in advanced.

shivmahesh, 2021-10-12:
grbl controllar error message: no data from com port after connect,expecting grbl version string.am using controllar 3.6.1 and grbl version 1.1