robottini/grbl-servo #7: How to change D11 to another pin?

stajor81, 2016-09-28:
Hi!
How to change D11 to another pin? D10 or D09 for example.
Thx!

robottini, 2016-09-28:
Hi, it isn't easy.
You can refer to GRBL documentation on github.
Bye.

stajor81, 2016-09-28:
How can I use Z-axis endstops on D11 and servo? Is there a simple solution?

cawoodm, 2017-08-22:
Since D11 is a Z-stop in vanilla GRBL I assume that you @robottini (grbl-servo) chose it arbitrarily for the servo? If so it was a poor choice since it means no CNC running grbl-servo cannot use stop/limit switches due to interferencce.

Better would be D04 since that is used for controlling the Z-Axis and, IMO, most CNC projects employing a Servo would use it for the Z axis (plotter pen up scenario).