robottini/grbl-servo #13: Configuration on MEGA 2560

earincon198, 2017-07-21:
Hello!
I have an arduino Mega 2560, I would like to know the correct pin to control the servo. I am already able to control my machine´s stepper motors for X and Y axis but not my servo. I changed the CPU map to #define CPU_MAP_ATMEGA2560 in the config.h file and connected my steppers to pins 25,26,30 and 31, and the steppers worked fine, then, I tried to connect my servo to all the PWM pins in my Arduno but it doen't work. I will apreciate any help, Thank you!

slimebdx, 2017-12-29:
hello i have the same problem , and i can't find the solution

lokeshraj23, 2018-02-17:
Hello !

I connected my servo to digital pin 7 and it worked for a while but after that it did not respond to M03 Sxxx. I did check my servo to see if it burnt out but nope it didnt.

ueline, 2021-08-16:
@earincon198 
I have a problem. I don´t get any respond from the serial monitor. 
I have the following equipment: 
-	Arduino Mega 2560 R3
-	CNC Shield with 2x A4988 Stepper Motor Driver
May I need to change the config.h file?

alexc57, 2022-11-09:
Same problem, x et y moved but nothing on D11. the servo doesn't moved.... any solution
grbl 1.1g on mks base v1.6 (atmega 2560)
thks in advanced