robottini/grbl-servo #17: Servo problem

Narcis-Daniel, 2018-02-22:
Hello 
I`m havin a problem with the servo
I connect the servo corectly (5v  gnd and Z+ (D11)) but when i try to use it or when it`s used( M03 Sxx) the servo moves a little bit and then everything stops similiar to what happens when the limits are used
I have to disconnect the arduino and plug it in again to work.
It`s like the Z limit is triggered
Please help
I am using cnc shield v3.

marsaiz, 2019-08-14:
Hola, no conectes el servo directamente al arduino o a la placa shield. Compra un regulador de voltaje LM7805 y alimenta el servo desde una fuente externa. El problema es el voltaje del pin que no llega a 4,8v. el mínimo para que funcione cualquier servo.
Saludos

wolfcatworkshop, 2021-03-14:
I'm going to translate @marsaiz above in case it's helpful to anyone else:

"Hi, don't connect the servo directly to the arduino or the shield. Buy a voltage regulator LM7805 and power the 
servo with an external source. The problem is the voltage pin not reaching 4.8 volts which is the minimum a servo
needs to work. "

Also, I was having the same problem and just replacing the servo did it for me. But there was definitely a problem with the
servo drawing too much power as every time I plugged in the servo it would disconnect.

KentNeal, 2021-11-19:
> I am not sure but after connect it to external source servo stop reacting on command at all probably I need som jumpers or something?

I burned a servo in frustration. The recommendation above is correct. External power and 5vdc regulator for the servo. Signal to z+

RomanIvanovMinsk, 2021-11-19:
Ok lads the answer probably simple - probably not. Recommendation above is correct BUT you have to connect ground external source power to the ground of CNC shield (in my example). Hope it save some tiem for other newbies that startr create pen-drawing-machine.