robottini/grbl-servo #20: Replace servo motor with a laser diode

panosss, 2019-01-25:
I replaced the servo motor with a laser diode, 5mW, [this one](https://www.seeedstudio.com/document/HLM1230.pdf).
But when the diode lights on (with M03 S255, or any number) it does not come off with M03 S1.
Or M05. 
So I have to modify the code to switch the diode on and off.
As a switch I use a 2N2222A transistor.

robottini, 2019-01-25:
Hi you have to use the vanilla GRBL, not this one. See https://github.com/grbl/grbl

panosss, 2019-01-25:
(thanks for the so quick reply!)
The vanilla GRBL is for bipolar stepper motors. 
There must be some way to use your code with a laser diode.

panosss, 2019-01-29:
I can't use the vanilla version of GRBL because it is for bipolar stepper motors.
I have unipolar.