robottini/grbl-servo #28: servo motor move only once

imfromkr, 2022-09-15:
It's been more than a month since this problem occurred. But I still haven't found a solution, so I'm asking for help here.
I purchased the xy-plotter from the aliexpress site.
When the command is given to the servo (m3 S40, m3 S90, etc.), it operates to the maximum angle and then there is no response. On YouTube, actions such as m3 S1000 and m5 work well, but my xy-plotter does not work with commands such as m3 S1000 and m5. However, arduino does not know what the problem is because the OK-sign appears every time you give a command.
I've seen some people who have problems like me while looking for solutions on Google or YouTube, but there's no solution to their posts, so I'm writing this myself. If you are aware of this problem or have solved it, please leave a comment. I will check the community whenever I have time.

If you have any other questions, I will answer them.

Below is the setting that I am using on YouTube.
$0=10 Step pulse, microseconds
$1=25 Step idle delay, milliseconds
$2=0 Step port invert, mask
$3=0 Direction port invert, mask
$4=0 Step enable invert, boolean
$5=0 Limit pins invert, boolean
$6=0 Probe pin invert, boolean
$10=3 Status report, mask
$11=0.010 Junction deviation, mm
$12=0.010 Arc tolerance, mm
$13=0 Report inches, boolean
$20=0 Soft limits, boolean
$21=0 Hard limits, boolean
$22=0 Homing cycle, boolean
$23=0 Homing dir invert, mask
$24=25.000 Homing feed, mm/min
$25=500.000 Homing seek, mm/min
$26=250 Homing debounce, milliseconds
$27=1.000 Homing pull-off, mm
$30=1000 Max spindle speed, RPM
$31=0 Min spindle speed, RPM
$32=0 Laser mode, boolean
$100=80.000 X steps/mm
$101=80.000 Y steps/mm
$102=80.000 Z steps/mm
$110=3000.000 X Max rate, mm/min
$111=3000.000 Y Max rate, mm/min
$112=3000.000 Z Max rate, mm/min
$120=3000.000 X Acceleration, mm/sec^2
$121=3000.000 Y Acceleration, mm/sec^2
$122=1000.000 Z Acceleration, mm/sec^2
$130=310.000 X Max travel, mm
$131=450.000 Y Max travel, mm
$132=200.000 Z Max travel, mm

aravindnc, 2023-01-30:
Same issue here. Pen is going up then there is no response.

aravindnc, 2023-01-30:
Solved issue by providing aux 5V power to the servo motor.