robottini/grbl-servo #30: Servo motor Arduino

SANTILIZER, 2023-06-15:
Hello every one I am new to Arduino and I am facing the following problem,
I have a problem with one of my servo motor (mg996R) it has the following Problems :
1- It is very shaky when I fix it to a certain angle for example if i tell it to go to angle 110 it will go there but it cant stand still 
2- After a certain time around 10 to 20 second (not always fixed time it can be below or above)  it  neglects all orders and starts moving on its own trying to break it is limited movement . meaning that if the motor is set on 110 angle suddenly it will start spinning to reach 0 angle and keeps the spinning  although it cant go beyond that ( even manually ) . Uploading a new Arduino code will not fix it nor cutting the power supply for the servos (it will start spinning once the power is back), it can only be fixed by cutting the supply for the servos then upload a new code to the Arduino.
I wish to know what is the cause if this issue because I cant find a solution to it any where ( and I did glue it to my intended part so simply changing it is quiet hard)

robottini, 2023-06-16:
Hi, I don't know how you provide the power to the servo. You can't power the servo from the Arduino. You have to provide the power supply externally from the Arduino, Very important is wire the external power supply GND with the Arduino GND.

robottini, 2023-06-18:
Hi, usually the 90% of the problems with the erratic servo is a power supply problem. Can you post the specs of your power supply and a picture of the wiring between the power supply, the servo and the arduino?

overland41, 2024-08-05:
good morning I had already written but perhaps I didn't see the answer I built one of the plotters on thingiverse and I installed the GRBL robots downloaded here on Arduino Uno and CNC Shield V3, the axes move and the machine works but the servo motor doesn't move neither with a g code file nor by typing on ugs M3 S90 OR M5
 TO GIVE AN ORDER TO LOWER OR RAISE, THE SERVO TENDS TO HEAT A LOT AND THEN IT'S AS IF IT NO LONGER WORKS THE SERVO IS AN SG90 AND IS FREE FROM OBSTACLES AND THE UGS 2.1.8 VERSION FOR 64 BIT, I try a lot I change the servo etc but nothing, I connect the servo with the red 5V wire, the brown GND and the yellow Z+, however it doesn't make any movement but it heats up a lot, perhaps GRBL is not compatible with Windows 11 64 bit, can you tell me if there is a solution, thanks
Invia commenti

overland41, 2024-08-05:
The whole machine is powered by a AC 100-240V 50-60Hz 0.3A output 2A power supply

robottini, 2024-08-05:
Hi, It seems you power the servo in a correct way. Sorry, I can't help you

overland41, 2024-08-05:
yes thanks but I just wanted to know this grbl that you released are you absolutely sure that it is compatible for this plotter and for the arduino uno board and cnc Shield because I have the impression that perhaps there is an anomaly in the code and from 2015 but I do not want to create other questions as many complain that downloaded this grbl the servo does not work maybe if in the future you can update it because even the operating systems and the arduino chips perhaps are no longer good for 2024 bye
Inviato da iPhone

robottini, 2024-08-08:
You can try other version of servo-GRBL: 
https://github.com/ufficioprogettiperduti/Scalable-Pen-Plotter
https://github.com/cprezzi/grbl-servo